commonsguy/cwac-cam2 #323: LG K8 Preview stretched while recording video

mbernr, 2017-05-15:
This is what it looks like on my LK K8 (Android 6.0):

Golfball before recording:
[image]

Golfball during recording: 
[image]

The actual output file looks fine.

commonsguy, 2017-05-15:
That appears to be the same as issue #288, so I am going to close this as a duplicate. 

If you could confirm for me the specific model (the LG K8 seems to have several), I may pick one up to do some specific testing on it. Is yours the K350K, the K371, or perhaps something else?

closed 2017-05-15 by commonsguy

mbernr, 2017-05-16:
It is the LG-K350N

commonsguy, 2017-05-16:
OK, thanks!