commonsguy/cwac-cam2 #324: TransformException: java.util.zip.ZipException: duplicate entry: com/android/mms/exif/ExifInterface$GpsDifferential.class

nehal07buddy, 2017-06-08:
After adding cam2 library i am not able to build the app and I am getting this error:

Error:Execution failed for task ':app:transformClassesWithJarMergingForDebug'.
> com.android.build.api.transform.TransformException: java.util.zip.ZipException: duplicate entry: com/android/mms/exif/ExifInterface$GpsDifferential.class

Please help.

commonsguy, 2017-06-08:
What other libraries do you have in your project?

nehal07buddy, 2017-06-08:
compile fileTree(include: ['*.jar'], dir: 'libs')
    compile fileTree(dir: 'libs', include: ['*.jar'])
    //    compile 'com.squareup.leakcanary:leakcanary-android:1.4-beta2'
    compile files('libs/commons-net-3.5.jar')
    compile files('libs/org.apache.http.legacy.jar')
    compile files('libs/PayU-release.jar')
    compile files('libs/YouTubeAndroidPlayerApi.jar')
    compile files('libs/gdata-contacts-3.0.jar')
    compile files('libs/gdata-client-1.0.jar')
    compile files('libs/gdata-core-1.0.jar')
    compile('com.sdk.beewise:beewisesdk:1.4') {
        exclude module: 'junit'
    }
    compile 'com.android.support:multidex:1.0.1'
    compile('com.google.code.gson:gson:2.6.2') {
        exclude group: 'com.google.gdata'
    }
    compile('ch.acra:acra:4.7.0') {
        exclude group: 'org.json'
    }
    compile 'com.balysv:material-ripple:1.0.1'
    compile 'de.hdodenhof:circleimageview:1.3.0'
    compile 'com.android.support:multidex:1.0.1'
    compile 'com.android.support:multidex:1.0.1'
    compile 'com.wang.avi:library:1.0.5'
    compile 'com.google.guava:guava:19.0-rc2'
    compile 'com.github.d-max:spots-dialog:0.4@aar'
    compile 'com.iceteck.silicompressorr:silicompressor:1.1.0'
    compile 'id.zelory:compressor:1.0.2'
    compile 'com.github.jakob-grabner:Circle-Progress-View:v1.2.9'
    compile 'com.github.ParkSangGwon:TedPicker:v1.0.10'
    compile 'com.facebook.android:facebook-android-sdk:4.+'
    compile 'com.cloudinary:cloudinary-android:1.4.2'
    compile 'com.segment.analytics.android:analytics:4.+'
    compile 'com.squareup.picasso:picasso:2.5.2'
    compile 'com.jakewharton:butterknife:8.1.0'
    compile 'com.squareup.retrofit2:retrofit:2.1.0'
    compile 'com.squareup:otto:1.3.8'
    compile group: 'com.zendesk', name: 'sdk', version: '1.9.1.1'
    compile 'com.balysv:material-ripple:1.0.1'
    compile 'de.hdodenhof:circleimageview:1.3.0'
    compile 'com.wang.avi:library:1.0.5'
    compile 'com.google.guava:guava:19.0-rc2'
    compile 'com.android.support:multidex:1.0.1'
    compile 'com.github.d-max:spots-dialog:0.4@aar'
    compile 'com.iceteck.silicompressorr:silicompressor:1.1.0'
    compile 'id.zelory:compressor:1.0.2'
    compile 'com.kbeanie:multipicker:1.1.3@aar'
    compile 'com.github.jakob-grabner:Circle-Progress-View:v1.2.9'
    compile 'com.android.support:recyclerview-v7:24.0.0'
    compile 'com.android.support:cardview-v7:24.0.0'
    compile 'com.android.support:appcompat-v7:24.0.0'
    compile 'com.android.support:design:24.0.0'
    compile 'com.google.android.gms:play-services:9.8.0'
    compile 'com.android.support:support-v4:24.0.0'
    compile 'com.github.ParkSangGwon:TedPicker:v1.0.10'
    compile 'com.facebook.android:facebook-android-sdk:4.+'
    compile 'com.cloudinary:cloudinary-android:1.4.2'
    compile 'com.segment.analytics.android:analytics:4.+'
    compile 'com.squareup.picasso:picasso:2.5.2'
    compile 'com.jakewharton:butterknife:8.1.0'
    compile 'com.squareup.retrofit2:retrofit:2.1.0'
    compile 'com.squareup.retrofit2:converter-gson:2.0.2'
    compile 'com.squareup:otto:1.3.8'
    compile 'com.github.delight-im:Android-AdvancedWebView:v3.0.0'
    compile 'com.google.android.gms:play-services-auth:9.8.0'
    compile 'com.google.android.gms:play-services-gcm:9.8.0'
    compile 'com.beardedhen:androidbootstrap:2.3.0'
    compile 'com.webengage:android-sdk:1.+'
    compile 'com.google.android.gms:play-services-analytics:9.8.0'
    compile 'com.droidninja:filepicker:1.0.8'
    compile 'com.google.code.gson:gson:2.8.0'
    compile 'com.razorpay:checkout:1.2.1'
    compile 'com.android.support.constraint:constraint-layout:1.0.0-beta4'
    compile 'com.google.android.gms:play-services-tagmanager:9.8.0'
    compile 'com.commonsware.cwac:cam2:0.7.+'
        testCompile 'junit:junit:4.12'
    apt 'com.jakewharton:butterknife-compiler:8.1.0'

nehal07buddy, 2017-06-08:
These are the libraries I am using

commonsguy, 2017-06-08:
Um, that is a preposterously long list.

Regardless, one of them added in the same EXIF-handling code from Google that I did.

Issue #325 is now tracking some work to refactor my copy of those classes, to help reduce the odds of a collision like this. In the meantime, either stop using my library or find out which other library has the EXIF classes and stop using that library.

closed 2017-06-08 by commonsguy

nehal07buddy, 2017-06-08:
Can you please tell me which library is causing the issue . If possible I can remove that library for now and continue with my development.

commonsguy, 2017-06-08:
> Can you please tell me which library is causing the issue 

I have no way to know, just by looking at their names. I can tell you that it is unlikely to be any of the `com.android` ones. The only way that I know of to identify who has those classes is to examine the contents of each library and see if the classes show up there.

nehal07buddy, 2017-06-08:
Okay. Thanks for the help. I ll check.

nehal07buddy, 2017-06-08:
I checked the libraries and I am getting this issue because of TedPicker Library. Is there any way I can resolve it from my side to fix it and be able to use both the libraries? If possible please let me know.

commonsguy, 2017-06-08:
> Is there any way I can resolve it from my side to fix it and be able to use both the libraries?

You could play around [with `pacakgingOptions`](https://google.github.io/android-gradle-dsl/current/com.android.build.gradle.internal.dsl.PackagingOptions.html) and see if there is a way to use that to only include one copy of those classes. I have never used `packagingOptions` for this, and so I have no idea if it can be used in this scenario.

Otherwise, there is no simple way to address this on your side.

nehal07buddy, 2017-06-08:
Ok Sure. Thanks. I ll look into possible solutions.