commonsguy/cwac-cam2 #326: Need to match preview size with current recording size

ssherk, 2017-06-16:
Both Google CTS Verifier and Android stock camera application ensures the preview and recording size is matched. 

Due to mismatch in resolutions, the recorded video capture is corrupted. Can you fix this ASAP as  
Akario Backline application is using this library for embedding camera and video captures as part of their securing messaging solution for healthcare.

As quick fix could you add our product to the constraints list to prevent the mismatch?

Constraints:

    add(new Builder()
      .manufacturer("Spectralink")
      .highCamcorderProfile(CamcorderProfile.QUALITY_1080P)
      .previewRFCSizeWhitelist(new Size(1920,1080))
      .pictureRFCSizeWhitelist(new Size(1920,1080))
      .supportsCameraTwo(false)
      .build());

commonsguy, 2017-06-16:
> Both Google CTS Verifier and Android stock camera application ensures the preview and recording size is matched.

That's not necessarily possible. Not every device has the Play Store and other Google proprietary apps; such devices do not need to comply with the CTS. A manufacturer is welcome to have a disjoint set of preview resolutions and video-recording resolutions.

(though if you have a link to the relevant CTS test, that would be interesting to see...)

> Can you fix this ASAP as
Akario Backline application is using this library for embedding camera and video captures as part of their securing messaging solution for healthcare.

First, there is a reason why the version number of this library starts with 0 &mdash; it is still very much a work-in-progress.

Second, this is a "spare time" library for me. Nobody is paying me to write this. That's one of the reasons why this is still a 0.x. In the best-case scenario that I see right now, it will be 2018 before I take it to a 1.0.

I would be happy to work out a support contract with the appropriate firm (Akario? DrFirst?). Contact me at consulting@commonsware.com.

That portion of this issue is a duplicate of #288. That happens to be relatively high on my list, in that I am going to spend some time on issues that require a full retest (as this one will) leading to a new minor release. However, that's still probably a month or two out at my current pace.

> As quick fix could you add our product to the constraints list to prevent the mismatch?

I don't have a problem with adding a constraint. But, can we do something narrower than `manufacturer("Spectralink")`? While it looks like they only have one Android model at the moment, I would hate to assume that these settings are proper for any future devices. Can you check the `Build` values and come up with something that scopes it to the specific product that you are using?

Leaving this open for the constraint, but see #288 for the longer-term fix.

ssherk, 2017-06-16:
Thanks Mark,

I wasn't making a case cwac needs to comply with CTS rules. I was just pointing this out as our devices has gone through the CTS testing and we are seeing this issues in the field with this application Akario.

I have no insight to the relation between Akario Backline and the use of this library. I will need to figure
out the logistics of this in order for them to pick up added constraints.

Thanks for adding the constraints to the list. I think this would be a good first step / quick fix.

The product is 8744 - getprop for product:    [ro.product.name]: [cyclops_6dq]

      .manufacturer("Spectralink")
      .product("cyclops_6dq")
      .highCamcorderProfile(CamcorderProfile.QUALITY_1080P)
      .previewRFCSizeWhitelist(new Size(1920,1080))
      .pictureRFCSizeWhitelist(new Size(1920,1080))
      .supportsCameraTwo(false)
      .build());

commonsguy, 2017-06-16:
> I wasn't making a case cwac needs to comply with CTS rules. I was just pointing this out as our devices has gone through the CTS testing and we are seeing this issues in the field with this application Akario.

My point was more that *I* cannot assume that there will always be a preview/recording resolution match. And I am expecting this change to break lots of things, which is why I have been putting it off.

 > I have no insight to the relation between Akario Backline and the use of this library

Um, that runs counter to the original issue report.

> The product is 8744 - getprop for product: [ro.product.name]: [cyclops_6dq]

Are you sure that `Build.PRODUCT` is identical to `ro.product.name`? I work at the app level, and I have never checked `getprop` values to see whether they correspond 1:1 to `BuildConfig` equivalents. I cannot exactly buy one of these off the shelf to test it myself.

ssherk, 2017-06-16:
I miss spoke,  I know that Akario Backline is using this library.
Yes I'm sure they correspond. I've tested this change by loading/running on the device.

ssherk, 2017-06-22:
Thanks for the change!