commonsguy/cwac-cam2 #334: result image is smaller than preview

itaybia, 2017-08-07:
the resulting image saved to the supplied URL is smaller (cropped) than what is seen on screen in the preview.
taken on a nexus 10, Android 5.1.1

                File temp = new File(getExternalCacheDir(), "temp_" + curTimeSec.toString() + ".jpg");
                captureFileName = temp.getAbsolutePath();
      		Intent intent = new CameraActivity.IntentBuilder(MyActivity.this)
                .skipConfirm()
                .requestPermissions()
                .facing(Facing.BACK)
                .to(temp)
//                .debug()
                .zoomStyle(ZoomStyle.NONE)
                .flashModes(Arrays.asList(FlashMode.OFF))
                .build();
                startActivityForResult(intent, REQUEST_CAMERA);

commonsguy, 2017-08-07:
There is no requirement that the preview match the picture. The aspect ratio of the preview precludes an exact match in most cases.

closed 2017-08-07 by commonsguy

itaybia, 2017-08-07:
can you please explain? how can a user tell what he is actually taking a picture of?
let's say he sees on screen his keyboard from "q" on the left to "p" on the right, but when he takes the picture, the actual result is from "r" to "i".
isn't it possible to restrict the preview to the resulting image by way of cropping/black bar addition? or by changing the aspect ratio to match the output?
the native camera gives an exact match.

at the moment, it will look quite strange for a user that what he sees is not what he gets.

commonsguy, 2017-08-07:
> isn't it possible to restrict the preview to the resulting image by way of cropping/black bar addition? or by changing the aspect ratio to match the output?

Yes. I had that originally (in the CWAC-Cam library). Then developers complained that I wasn't filling the preview screen. So I switched to filling the preview screen. Now developers like you complain that I am filling the preview screen and so the preview does not match the picture. And, frankly, I am tired of all of it.

I recommend that you not use this library, as I will be retiring it soon. I will be steering developers towards Fotoapparat and CameraKit-Android as alternatives.

itaybia, 2017-08-07:
i get that. didn't mean that you had to change it for me...
just wanted to understand why this is like that. and after reading:
https://stackoverflow.com/questions/11350085/custom-camera-andorid-preview-picture-is-different-from-show-picture
i now understand that the preview and capture are 2 separate mechanisms on android. too bad there's no reasonable way to sync them.
i guess i'll remove the camera from my app for now, as i too don't want the hassle of the various issues with it.
thanks for the recommendations.

commonsguy, 2017-08-07:
> didn't mean that you had to change it for me...

You are not the first developer who asked. There are a couple of closed issues here with the same request.