commonsguy/cwac-cam2 #336: Recording video fails in some conditions on devices not listed in CameraConstraints

shtolik, 2017-08-16:
When trying to record video (and some combination of parameters for taking picture) app crashes with following exception:
```
 Process: com.commonsware.cwac.cam2.playground:cwac_cam2, PID: 26477                                                                                                java.lang.RuntimeException: Unable to start activity 
ComponentInfo{com.commonsware.cwac.cam2.playground/com.commonsware.cwac.cam2.VideoRecorderActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean com.commonsware.cwac.cam2.CameraConstraints.supportsFFCVideo()' on a null object reference
```
It happens because not everywhere CameraConstraints is check for being null.
I've added checks in such places and going to create a pull request.

commonsguy, 2017-08-16:
Please do not provide a pull request. Please [follow the instructions](https://github.com/commonsguy/cwac-cam2/blob/master/.github/CONTRIBUTING.md) and provide enough details for me to reproduce the problem.

Or, better yet, switch to a different camera library (e.g., CameraKit-Android), as I will be discontinuing this one in the coming weeks.

shtolik, 2017-08-16:
Is zip file also generated for demo-playground? Is it in app folder on memory card of a phone? I'll try to find it. 

Ok I won't file the pull request but changes are [here](https://github.com/shtolik/cwac-cam2/commit/4c0fdda7657a2adea082a2abe775ff93348f27d3) if you'd need it.

I was only evaluating the lib yet, so thanks for warning about discontinuation.

commonsguy, 2017-08-16:
> Is zip file also generated for demo-playground?

No, sorry, that's only the `demo/` project.

What I really need is the product identifying details ("the `Build.MANUFACTURER` and `Build.PRODUCT` values for the hardware on which you are experiencing the problem").

> Ok I won't file the pull request but changes are here if you'd need it.

The problem with your pull request is that it is not solving the real problem. There should be few if any circumstances under which `CameraConstraints.get()` returns `null`. Hence, I would need to determine *why* that is `null`, not just ignore it when it is `null`.

shtolik, 2017-08-16:
I've already updated the necessary details in a first message, sorry for not mentioning it in a previous comment (Samsung G389F. Build.PRODUCT:MMB29K.G389FXXU1AQG1). 

And fixing `CameraConstraints.get()` returning null was my first approach, and I tried to generate default Constraint if no constraints are matched. 
```
if (match == null){
		  DEVICE_CONSTRAINT = new Builder()
				  .manufacturer("Unknown")
				  .build();
	  }

```
That helped, but then I saw that `CameraConstraints.get()` is checked for `null` in many places and decided that it was intended behaviour, missed in couple of places.

commonsguy, 2017-08-16:
> I've already updated the necessary details in a first message

My apologies for forgetting about that with my previous comment. It is unlikely that I will be able to acquire an SM-G389F here in the US any time soon to try to reproduce this. But, I will keep an eye out for one. Thanks!