commonsguy/cwac-cam2 #338: Autofocus failed runtime exception

VBegoyan, 2017-11-10:
Hi, i am experiencing with this issue: 
Received device exception during capture call: 
                                                                     java.lang.RuntimeException: autoFocus failed
                                                                         at android.hardware.Camera.native_autoFocus(Native Method)
                                                                         at android.hardware.Camera.autoFocus(Camera.java:1300)
                                                                         at android.hardware.camera2.legacy.LegacyFocusStateMapper.processRequestTriggers(LegacyFocusStateMapper.java:205)
                                                                         at android.hardware.camera2.legacy.RequestThreadManager$5.handleMessage(RequestThreadManager.java:824)
                                                                         at android.os.Handler.dispatchMessage(Handler.java:98)
                                                                         at android.os.Looper.loop(Looper.java:234)
                                                                         at android.os.HandlerThread.run(HandlerThread.java:61)

Device is: Sony D6503, Android 6.0.1, Api 23

VBegoyan, 2017-11-10:
Forget to mention, that i am working on Camera2

commonsguy, 2017-11-10:
As is noted on the project home page, this project is discontinued.
Please use another library, such as Fotoapparat or CaneraKit-Android.